dronecan/gui_tool #6: Type error on load

WickedShell, 2022-01-19:
After installing with sudo I get the following error:

``` python
2022-01-19 14:44:05,439 INFO dronecan_gui_tool.main Init complete, invoking the Qt event loop
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 141, in pixmap
    self.paint(QPainter(pm), QRect(QPoint(0, 0), size), mode, state)
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 135, in paint
    self.painter.paint(
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 65, in paint
    self._paint_icon(iconic, painter, rect, mode, state, opt)
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 112, in _paint_icon
    painter.setFont(iconic.font(prefix, draw_size))
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 311, in font
    font.setPixelSize(size)
TypeError: setPixelSize(self, int): argument 1 has unexpected type 'float'
QPainter::end: Painter ended with 2 saved states
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 141, in pixmap
    self.paint(QPainter(pm), QRect(QPoint(0, 0), size), mode, state)
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 135, in paint
    self.painter.paint(
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 65, in paint
    self._paint_icon(iconic, painter, rect, mode, state, opt)
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 112, in _paint_icon
    painter.setFont(iconic.font(prefix, draw_size))
  File "/usr/lib/python3.10/site-packages/qtawesome/iconic_font.py", line 311, in font
    font.setPixelSize(size)
TypeError: setPixelSize(self, int): argument 1 has unexpected type 'float'
QPainter::end: Painter ended with 2 saved states
```

The painter tracebacks repeats a number of times that I trimmed here for brevity. The icons are all missing in the GUI as well.

davidbuzz, 2022-02-03:
https://doc.qt.io/qt-5/qfont.html#setPixelSize     says that font.setPixelSize(size) expects its argument to be an integer in the QT bindings themselves,  and python package 'qtawesome'  => iconic_font.py isn't  properly ensuring that...   is that right?

davidbuzz, 2022-02-03:
either way... tridge's mention suggests this might now be fixed in master...  can u retest, @WickedShell  ?

closed 2022-02-03 by davidbuzz

WickedShell, 2022-02-03:
@davidbuzz I can't retest, because in the effort to fix the installer, the installer is now completely broken for me. If I do a pip install as sudo (which should not be needed, but seperate converastion), it succeeds without any listed error, but there isn't a valid program anywhere on disk to run. If I try to install it as user then it just bails due to invalid permissions.

``` bash
$ sudo pip install install git+https://github.com/DroneCAN/gui_tool@master
[sudo] password for wickedshell: 
Collecting git+https://github.com/DroneCAN/gui_tool@master
  Cloning https://github.com/DroneCAN/gui_tool (to revision master) to /tmp/pip-req-build-joh_rwof
  Running command git clone -q https://github.com/DroneCAN/gui_tool /tmp/pip-req-build-joh_rwof
  Running command git submodule update --init --recursive -q
Collecting install
  Downloading install-1.3.5-py3-none-any.whl (3.2 kB)
Requirement already satisfied: setuptools>=18.5 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (59.1.1)
Requirement already satisfied: dronecan>=1.0.0 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (1.0.5)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (3.5)
Requirement already satisfied: qtawesome>=0.3.1 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (0.3.3)
Requirement already satisfied: qtconsole>=4.2.0 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (5.2.2)
Requirement already satisfied: pyyaml>=5.1 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (5.4.1)
Requirement already satisfied: easywebdav>=1.2 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (1.2.0)
Requirement already satisfied: numpy in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (1.22.1)
Requirement already satisfied: pyqt5 in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (5.15.6)
Requirement already satisfied: traitlets in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (5.1.1)
Requirement already satisfied: jupyter-client in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (7.1.1)
Requirement already satisfied: ipykernel in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (6.7.0)
Requirement already satisfied: pygments in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (2.11.2)
Requirement already satisfied: qtpy in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (2.0.0)
Requirement already satisfied: pyqtgraph in /usr/lib/python3.10/site-packages (from dronecan-gui-tool==1.0.0) (0.12.3)
Requirement already satisfied: requests in /usr/lib/python3.10/site-packages (from easywebdav>=1.2->dronecan-gui-tool==1.0.0) (2.26.0)
Requirement already satisfied: six in /usr/lib/python3.10/site-packages (from qtawesome>=0.3.1->dronecan-gui-tool==1.0.0) (1.16.0)
Requirement already satisfied: pyzmq>=17.1 in /usr/lib/python3.10/site-packages (from qtconsole>=4.2.0->dronecan-gui-tool==1.0.0) (22.3.0)
Requirement already satisfied: ipython-genutils in /usr/lib/python3.10/site-packages (from qtconsole>=4.2.0->dronecan-gui-tool==1.0.0) (0.2.0)
Requirement already satisfied: jupyter-core in /usr/lib/python3.10/site-packages (from qtconsole>=4.2.0->dronecan-gui-tool==1.0.0) (4.9.1)
Requirement already satisfied: ipython>=7.23.1 in /usr/lib/python3.10/site-packages (from ipykernel->dronecan-gui-tool==1.0.0) (8.0.1)
Requirement already satisfied: tornado<7.0,>=4.2 in /usr/lib/python3.10/site-packages (from ipykernel->dronecan-gui-tool==1.0.0) (6.1)
Requirement already satisfied: nest-asyncio in /usr/lib/python3.10/site-packages (from ipykernel->dronecan-gui-tool==1.0.0) (1.5.4)
Requirement already satisfied: matplotlib-inline<0.2.0,>=0.1.0 in /usr/lib/python3.10/site-packages (from ipykernel->dronecan-gui-tool==1.0.0) (0.1.3)
Requirement already satisfied: debugpy<2.0,>=1.0.0 in /usr/lib/python3.10/site-packages (from ipykernel->dronecan-gui-tool==1.0.0) (1.5.1)
Requirement already satisfied: stack-data in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.1.4)
Requirement already satisfied: decorator in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (5.1.1)
Requirement already satisfied: black in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (21.12b0)
Requirement already satisfied: jedi>=0.16 in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.18.1)
Requirement already satisfied: backcall in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.2.0)
Requirement already satisfied: pickleshare in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.7.5)
Requirement already satisfied: prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0 in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (3.0.24)
Requirement already satisfied: pexpect>4.3 in /usr/lib/python3.10/site-packages (from ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (4.8.0)
Requirement already satisfied: parso<0.9.0,>=0.8.0 in /usr/lib/python3.10/site-packages (from jedi>=0.16->ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.8.3)
[... cut ...]
Requirement already satisfied: pure-eval in /usr/lib/python3.10/site-packages (from stack-data->ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.2.1)
Requirement already satisfied: executing in /usr/lib/python3.10/site-packages (from stack-data->ipython>=7.23.1->ipykernel->dronecan-gui-tool==1.0.0) (0.8.2)
Installing collected packages: tomli, install
  Attempting uninstall: tomli
    Found existing installation: tomli 2.0.0
    Uninstalling tomli-2.0.0:
      Successfully uninstalled tomli-2.0.0
Successfully installed install-1.3.5 tomli-1.2.3
```