dronecan/gui_tool #27: NodeStatus not received when custom DSDLs are loaded.

dernatsch, 2022-08-31:
When loading custom DSDL files in the startup dialog the NodeStatus monitor stops working. This is also true for an empty DSDL directory.

My analysis:
After specifying a folder for custom DSDLs the method `load_dsdl` is called again and it replaces the contents of the `DATATYPES` dict. Because the datatype objects contain closures they are generated anew. The problem arises when a new frame is received and the frames datatype and the NodeStatus datatype are compared. It seems the NodeStatus datatype used in the comparison originates from the first load of the DSDL files and the datatype of the received frame orinates from the `DATATYPES` dict. Because of the closures this comparison returns False even if both messages have the NodeStatus type.
There are multiple approaches to this problem that I can think of:
- Not overwriting the contents of the `DATATYPES` dict when it already loaded a definition.
- Basing the comparison of datatypes on the signature instead of the comparison of the python objects.
Pseudo:
```python
def __eq__(self, other):
    self.signature() == other.signature()
```

dtran11, 2022-10-15:
I see the same issue. Is this a problem with dronecan module or this gui project?

FelixEL, 2024-07-20:
this is a pydronecan problem. I confirm the same behavior where loading custom DSDL deactivates other definitions.