dronecan/gui_tool #35: CAN bus monitor lags and consumes excessive CPU resources when recording frames

jensenr30, 2023-06-15:
In the CAN bus monitor window, when I press the "record" button, as CAN frames build up in the log, the performance of the app gradually declines.  Refresh rates gradually slow to less than 1 screen refresh per second.  CPU usage increases to 5-6%, and memory usage seems to climb slowly without limit.  When the CAN bus monitor is not recording, the CPU usage drops to less than 0.05%

Linux 6.2.5-060205-generic #202303110831 SMP PREEMPT_DYNAMIC Sat Mar 11 13:57:40 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 23, i5-6000k, 32 GB DDR4 @ 3200 MT/s

If I find some free time one of these weekends, I would like to try contributing to this project by investigating these performance issues.  Which sources files should I look at to improve CAN bus monitor performance?

On my machine it seems that the memory usage grows without bound.  I'm thinking it would be nice to save most of the old CAN frame logs to disk to avoid hogging RAM.

I'm not sure what changes would reduce CPU usage.

thealienthing, 2023-07-06:
I'd like to second this. I'm experiencing the same issues on Windows 10 where the Bus Monitor is SUPER laggy. My CPU usage is at 12% on average. Memory is not too much of an issue but dumping messages from RAM occasionally would be really nice.

jensenr30, 2023-07-06:
The issue might be related to rendering the list of CAN messages.  If that is the case, maybe the performance could be improved by only rendering the messages that fit into the current view of the list.

thealienthing, 2023-07-16:
I was wondering the same thing. Suppose this is one of those moments where we're supposed to roll up our sleeves and do it lol.

David-OConnor, 2023-08-23:
I considered writing a tool like this in Rust; performance currently is not good.

thealienthing, 2023-09-08:
As a workaround, I find the best approach is to set up filters, like filter messages intended for a particular node, THEN click the capture button. Capture messages until you've got enough data. Then pause and then stop capturing. This approach allows me to keep the monitor open for long stretch without needing to close it before it seizes up 🤣