dronecan/gui_tool #94: Setup Tools Permission Denied to write to /usr/local/share/applications/DroneCAN GUI Tool.desktop on Linux

danapolsUM, 2025-06-24:
OS: Ubuntu 22.04.5

The removal of [sudo](https://github.com/dronecan/gui_tool/commit/7186245ab6678c13841e57795309d414fa1e35b8#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R43) in the Debian-based distribution installation causes the following error due to `/usr/local/share/application` needing sudo-level permissions to write to:
```
 Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      running dist_info
      creating /tmp/pip-modern-metadata-53uoiwmb/dronecan_gui_tool.egg-info
      writing manifest file '/tmp/pip-modern-metadata-53uoiwmb/dronecan_gui_tool.egg-info/SOURCES.txt'
      writing manifest file '/tmp/pip-modern-metadata-53uoiwmb/dronecan_gui_tool.egg-info/SOURCES.txt'
      running install_desktop
      /usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      error: [Errno 13] Permission denied: '/usr/local/share/applications/DroneCAN GUI Tool.desktop'
      [end of output]
```

A possible solution to not using sudo would be to instead store the `.desktop` file in `~/.local/share/applications`. I looked into this a little bit with the [install_freedesktop](https://github.com/welshjf/install_freedesktop) package currently being used, including trying to pass --user as a command line argument to the pip installation command, but it didn't seem to change anything. `install_freedesktop` appears to take arguments from [distutills.command.install](https://docs.python.org/3.11/distutils/apiref.html#module-distutils.command.install). The argument relevant to the installation destination of the `.desktop` file is `install_data`. However, this might cause the locations of other files to change as well, but I'm not sure.

WebDust21, 2025-07-15:
A crude bypass is to change the permissions on the destination folder.  In my case, it's trying to write to /usr/local/share/applications...so I just enabled write permissions on that folder to include "others".  By no means an ideal solution, but it DOES allow DroneCAN GUI to install.

[image]

Installed...oops, the shortcut that was causing that grief doesn't work...points to an invalid location ("/dronecan_gui_tool-1.2.28.data/scripts/dronecan_gui_tool")
Fortunately, "whereis dronecan_gui_tool" did pop out the correct answer: "dronecan_gui_tool: /home/[user]/.local/bin/dronecan_gui_tool".  Clicking on it -> it runs.

Hey, I didn't say this was the right way around the issues ;-)

WebDust21, 2026-04-28:
I don't recall, but seeing as I'm not a Python whiz (keep getting eaten by the snake, lol!), it _most likely_ was just the "python3 -m pip install" listed in the repo readme.

Got bigger issues with the DroneCAN GUI tool than that...particularly with the buggy SLCan implementation.  But that's for a different issue.  (In short: DroneCAN's internal Python SLCan implementation does not work with the Canable 2.0 in SLCan mode, "SLCan Ack Timeout".  _Barely_ works with the Canable 2.0 in "candlelight" mode.  Discovered that if I wrote some 'glue' to bridge the python-can SLCan implementation to pydronecan's CAN consumer, then I can easily use the Canable 2.0 in SLCan mode on DroneCAN.)